MCU-Innovations/HondaECU #14: prevent starting read/write when state prevents it

RyanHope, 2019-02-17:
prevent starting read/write when state prevents it

CodeFactoryFlo, 2019-03-20:
Hi Ryan,
great Tool, but i've good problems during boot of the ECU.
the ECUSTATE changes rapidly ON/OFF and it is not possible to read from ECU.

I've tried to read it on the bench with 3A Basetech Power Supply @ 13,0V and 0,22A
I'm Using an K+DCAN cable with FTDI Chip and OBD breakout cable 
(This Cable works for "Honda Flasher" with original FTDI USB Serial Drivers 2.12.16)
I've used the libusbK driver from the Zadig tool.

Is there maybe a driver issue? it seems the ecu is in a boot loop?
I've attached a video and my Pinout to the ECU.

[image]

[HondaECU_Keihin_D31_CBR1000RR07.zip](https://github.com/MCU-Innovations/HondaECU/files/2986972/HondaECU_Keihin_D31_CBR1000RR07.zip)

Regards
Florian

RyanHope, 2019-03-20:
Hey, I dont think this is a driver issue. And I don't think the ECU is
actually rebooting. As more people have started using my project, I have
struggled getting my code to play nice with all the different ftdi cable
variants out there. Basically, because of circuity differences, my
techniques for detecting if there is power on the kline dont always work,
which is why it keeps say its off when it probably isnt. My DIY kline
adapters behave one way, the vag com clones behave another way, the k+dcan
cables behave another way. I want to rip my hair out sometimes (haha). I
have almost a dozen different "methods" for detecting and some for some
people and some dont. I have a K+DCAN cable that does not work wit my
project right now, but I have had others say their K+DCAN cables work out
of the box. Hopefully I can figure out a technique that works for everyone.
This is a high priority for me as this affects a bunch of people right now.

CodeFactoryFlo, 2019-03-20:
> Hey, I dont think this is a driver issue. And I don't think the ECU is actually rebooting. As more people have started using my project, I have struggled getting my code to play nice with all the different ftdi cable variants out there. Basically, because of circuity differences, my techniques for detecting if there is power on the kline dont always work, which is why it keeps say its off when it probably isnt. My DIY kline adapters behave one way, the vag com clones behave another way, the k+dcan cables behave another way. I want to rip my hair out sometimes (haha). I have almost a dozen different "methods" for detecting and some for some people and some dont. I have a K+DCAN cable that does not work wit my project right now, but I have had others say their K+DCAN cables work out of the box. Hopefully I can figure out a technique that works for everyone. This is a high priority for me as this affects a bunch of people right now.
> […](#)
> -- Ryan Hope, PhD Psychology Department Carnegie Mellon University

Hi, thanks for your answer!
I've other types FTDI cables, do you have a recommendation for a specific FTDI chip?
Maybe I can help you with testing some? 

Regards
Florian.

bikedyno, 2019-03-20:
hi
i use a cheap chinese odb2 to usb , with simple sub harrness to honda diag
plug
use the  libusbK driver from the Zadig tool install driver for odb2
works perfectly

i built the interface ryan built , works perfect as well
i use the odb2 with sub harnness more pro looking setup

thanks pete
www.bike-dyno.com.au

CodeFactoryFlo, 2019-03-20:
> hi i use a cheap chinese odb2 to usb , with simple sub harrness to honda diag plug use the libusbK driver from the Zadig tool install driver for odb2 works perfectly i built the interface ryan built [...]

May you have a Link to the cable?

Thx Florian

CodeFactoryFlo, 2019-03-20:
Hi Pete,
may you have a link to the cable?

Thanks
Florian

bikedyno, 2019-03-20:
Hi
Yes
Will send diagram
Odb2 pinout also
I'll go through my arkive off diagrams
Asnd send to you to morrow

Thanks

RyanHope, 2019-03-20:
I am moving at the end of the month so I wont be too productive in the
next 2 weeks as a lot of my toys/tools are packed up. I have some
other cables as well and will do some more testing once I get settled
into the new place. The fact that your cable works with HondaFlasher
means I should be able to get it to work as well. These diagnostic
cables can be thought of as 2 components, the kline circuit which
merges the Tx and Rx lines and converts the 12v bike logic to 5v, and
then the ftdi chip which converts the serial kline communication to
usb. All of my issues lie in the kline circuit side of things I
believe. However, the ftdi chips do have an eeprom with some settings
and its possible this is what is causing some issues. For example, the
ftdi chip has a command to just read the pin states without having to
write to a pin. This code works perfect with all my diy circuits but
when I use a vag com clone and turn off the ECU, my code will detect
an off state for a second or 2 and then the circuit goes high again
even though there is no power on the ecu/bike side of things.

bikedyno, 2019-03-20:
Hi
Yes that is correct

will detect
an off state for a second or 2 and then the circuit goes high again

Is good it just works
I read a while ago . Some guy reprogram 1 cheap odb2

So i try'd it with Zadig tool and your software works perfectly

I have pinouts
For odb2 and honda sub leed also
I'll send

Thanks pete

bikedyno, 2019-03-20:
hi
pinout pictures for odb2 to honda mcu flasher

CodeFactoryFlo, 2019-03-21:
Hi, 
I've tested yesterday 3 different cables on Windows 7 and Windows 10 64Bit.
all of them have an FTDI FT232RL Chip and are programmable EEPROMs.
They are not working, as bikedyno said you need an old dumb Chinese VAG KKL (K-line) cable. :-)

Regards
Florian.

Gump35, 2019-04-19:
Hi Ryan,
Is this program working with Lonelec Hondaflasher cable?
[https://www.lonelec.co.uk/Motorbike-OBD-Diagnostic-Cables/Honda-ECU-Flasher-Motorbike/Honda-Motorbike-ECU-Flasher-Cable](url)

closed 2019-08-08 by RyanHope