MCU-Innovations/HondaECU #18: READ 2019 CRF250

BE514, 2019-05-14:
Cannot read 2019 CRF250 ecu?

RyanHope, 2019-05-14:
Well aware of this. Same issue on the CRF450. The ECU accepts all the normal read commands, accepts the read password, then stops responding. I believe there might be a baudrate switch or some sort of timing issue related to when the commands are sent. Haven't figured it out yet though... working on it.

BE514, 2019-05-14:
Ok thanks for the quick response. Is there anyway to read/write the
fuel/ignition trim tables for Mode 2 & 3 like your XDF shows for the 2017+
CRF450 model?

RyanHope, 2019-05-14:
I believe they are the same. When I tested the data settings stuff that was
on a 450 ecu. I do now have a 250 ecu in my possession so I could try it
out when I get a chance.

BE514, 2019-05-14:
I believe they are also, I'm not exactly sure how to read or write these
values to the ECU but let me know if I can help at all?

bikedyno, 2019-07-21:
Hi
Do You the xdf defination for this ECU, 
Turnerpro RT

THANKS
Pete

BE514, 2019-07-21:
No I do not have XDF. As far as I know no one has been able to read the bin file from these newer CRF ECU. 

Do you have a bin file?

Nutt53, 2019-07-30:
> ฉันเชื่อว่าพวกเขาเหมือนกัน เมื่อฉันทดสอบสิ่งที่การตั้งค่าข้อมูลที่อยู่ใน 450 ecu ตอนนี้ฉันมี 250 ecu ไว้ในความครอบครองของฉันแล้วดังนั้นฉันจึงสามารถลองใช้ได้เมื่อมีโอกาส
> [...](#)
> -- Ryan Hope, PhD Psychology Department Carnegie Mellon University

Hi bro 
please have your file crf250.bin.
qq.nutt53@gmail.com