MCU-Innovations/HondaECU #30: CBR 600rr 38770 mee 643

vasiliouv84, 2019-06-23:
I have this ECU but I can't work with your program.no live data no nothing.please update your program of you can.

vasiliouv84, 2019-06-25:
can you update the program for the CBR 600rr 2004 model and add any xdf of this model?

vasiliouv84, 2019-06-27:
cbr 600rr 2004 ecu bin file
[38770-MEE-643.zip](https://github.com/MCU-Innovations/HondaECU/files/3333487/38770-MEE-643.zip)

vasiliouv84, 2019-06-28:
I can't connect my ECU with your program.what I am doing wrong?

RyanHope, 2019-06-28:
What cable are you using?

vasiliouv84, 2019-06-28:
a vag com with ftdi ft232rl chip.it is the same cable that I use to extract
my ECU bin with honda ECU flasher a tool from an Italian guy.

vasiliouv84, 2019-06-30:
> What cable are you using?

any ideas what to do??

vasiliouv84, 2019-07-01:
I believe that you have to add the ECU Id at your program to work.if there is anything that I can do please tell me.

JabaRacing, 2019-07-02:
first you need to install libusbk for windows
after running the program going to file> devices it will show your cable as
a device option, if this is not happening it will not be possible to
communicate

vasiliouv84, 2019-07-02:
I have done this things but nothing.

CBRTricolor, 2019-07-02:
I’ve the same problem ☹
 

  <https://t.xink.io/Tracking/Impression/eiUAAIW_AADy7zwA0> 

Enviada: 2 de julho de 2019 18:55
Para: MCU-Innovations/HondaECU <HondaECU@noreply.github.com>
Cc: Subscribed <subscribed@noreply.github.com>

RyanHope, 2019-07-02:
Adding the ECU id has nothing to do with this.

vasiliouv84, 2019-07-02:
but as I have seen the ECU Id isn't at your program.if you add it maybe it will work.if not what else can we do to help?

vasiliouv84, 2019-07-06:
with the newest beta version I was able to connect to my ECU but the ECU state was going from on to unknown state.i was able to use the logger but I was unable to read or write the ECU of my bike.

bikedyno, 2019-07-21:
Hi
Do You the xdf defination for this ECU,
Turnerpro RT

THANKS
Pete

closed 2019-08-05 by RyanHope