MCU-Innovations/HondaECU #37: Help: ECUSTATE.Unknown

TrustyTristan, 2019-07-07:
Hi I am having a problem getting it to read my ECU (2008 CBR1000RR)

I am running:

- Windows 10.
- libusbK driver 3.0.7.0
- FT232R USB serial Adapter
- HondaECU v3.0.12

**From Bike -> Serial Interface**
Orange/White -> K Line
Black/White -> +12v
Green -> Ground

I have made the serial interface board like the diagram, but am using 470 and 510 Ohm resistors, and 2 * 1k Ohm (also tried just 1 * 1k Ohm)

Starting the program and plugging in the USB with the bike off:

- ECUSTATE.OFF

Turning on ignition

- LED flashing
- ECUSTATE.UNKNOWN

Turning on the bike

- LED flashing
- ECUSTATE.UNKNOWN

I have attached a debug log, saved from bike being, off, ignition on, bike started, bike off. 
[20190707-0943.txt](https://github.com/MCU-Innovations/HondaECU/files/3365161/20190707-0943.txt)
[image]

RyanHope, 2019-07-07:
Is your FTDI adapter set to 5v mode?

TrustyTristan, 2019-07-07:
> Is your FTDI adapter set to 5v mode?
> […](#)

Yes, I've also tried swapping it out with a different one (same model)

RyanHope, 2019-07-07:
Has it ever worked? Do you by any chance have an older version of
windows to test on or a different usb port? Just trying to confirm
it's a software issue and not a hardware issue.

TrustyTristan, 2019-07-07:
Totally understand.

Yeah the serial adapter was working fine for another project just yesterday.

I don't have an older version of windows unfortunately.

- I've tried all 3 USB ports.
- I've tried a different USB cable
- I've tried comparability mode for windows 8 and 7

Am I assuming correctly that it should read the ECU when the ignition is on?

Here is a picture of my breadboard, in case I messed up something on that, but I don't think that will be easy to follow.

[image]

RyanHope, 2019-07-23:
Did you ever get anywhere with this?

TrustyTristan, 2019-07-23:
No, I had someone else check my circuit for me to make sure that's right.

I was going to try a different version of Windows today. Which have you tested it on? Windows 7 or 8?

TrustyTristan, 2019-07-25:
I tried Windows 8.1 and that was worse, Might have been a bad install though...
[image]
[image]

I installed Windows 7 August 2018 after that and I had the same unknown problem.

I tried downloading again, and running different versions.

I verified the USB cable, FT232RL adapter, and USB ports all work.

So that makes me think it must be my circuit (resistors might be more important?)
Or the libusbK 3.0.7.0 driver
Or the adapter I have..?

RyanHope, 2019-07-25:
There are fake ftdi adapters out there. I just bought a batch of them to
make adapters and they are all fake. Have no eeprom, change change many
settings on them.

RyanHope, 2019-07-25:
On a side note, I just started working on my '15 Grom ECU again and I am
having LOTs of issues communicating with it. No issues with the 16+ Groms
or any 1000rr ECUS. This could all be related to my code.

TrustyTristan, 2019-07-25:
I think that's it >.<

Damnit... Trying to find one that has an EEPROM now...

RyanHope, 2019-07-26:
I am 99% sure all issue are related to fake FTDI chips. Seems like the fakes are getting worse and worse. I just wasted most of 2 days trying to diagnose some weird issues that 100% went way when I switched to a know genuine chip.

GeorgeTurbo, 2019-07-26:
Hello ! I'm a beginner. And I have a problem as well. I use the products of https://www.lonelec.co.uk/Motorbike-OBD-Diagnostic-Cables/Honda-ECU-Flasher-Motorbike/Honda-Motorbike-ECU-Flasher-Cable. !!! I want to know the installation process. From the beginning. We need to use more drivers. Thank you.

RyanHope, 2019-07-26:
I am in the process of making the driver install even simpler, stay tuned.

TrustyTristan, 2019-07-27:
IT WORKS RYAN!

I went and bought an expensive serial adaptor from the store instead of eBay, ...and it still didn't work :(

So in desperation I tried getting the exact resistor values.. And changing the K-Line (Bike side) to 1.2K ohm worked!
With both adaptors!

[image]

So lessons..
maybe resistor values are kinda important
maybe don't buy cheap adaptors..

Back on Windows 10 btw

GeorgeTurbo, 2019-07-27:
May I ask you something? BIN and xdfs files that have been tuned from Tunerpro RT. How do I put them in hondaecu v3.0.12? I downloaded the ZIP kit. Come, but don't use it. I use it as an exe set. Thank you.

RyanHope, 2019-07-27:
So with the 1k resistor the logic levels are probably too high on the rx pin of the ftdi chip... thanks for figuring this out.

closed 2019-07-27 by RyanHope

RyanHope, 2019-07-27:
Just wanted to thank you again for figuring this out @TrustyTristan. I just added a 220ohm resistor inline with the 1k and that resolved my communication issues with my old Grom ecu. Gonna have to go fix all the adapters I just made.