MCU-Innovations/HondaECU #38: Honda Small bike ecu compatibilty?

edwinopie, 2019-07-15:
hi,
can i use HOndaecu to read this ecus:

38770-K25-901 BeAT FI, BeAT FI CBS, Scoopy FI, Scoopy eSP
38770-K45-N42 New CBR 150R K45G
38770-K46-N01 Scoopy eSP (New) & Vario 110 eSP
38770-K15-902 CB150R StreetFire (Old)
38770-K18-902 Verza 150 FI
38770-K41-N01 Blade FI & Supra X 125 FI
38770-K15-921 New CB150R Streetfire K15G
38770-K84-901 CRF150
38770-K45-N01 CBR150R 2014-2016

30400-K25-600C1 BeAT POP eSP & Scoopy eSP
30400-K60-B11 Vario 125 eSP (B11)
30400-K59-A01 Vario 150 eSP
30400-K46-N31 New Vario 110 FI eSP
30400-K81-N62 Beat ESP
30400-K59-A71 New Vario 150 K59J

i tried to connect, success. datalog working, but when i try to read it show paswword failed.

any suggestion?

yvan746, 2019-07-21:
Hi ,
I have the same problem.
Which tool you use (FTDI or build Ryan tool?)?

edwinopie, 2019-07-21:
I use ftdi 232rl.. its a vagcom usb kkl..

edwinopie, 2019-07-21:
Which tool you are using? Can you show me?

bikedyno, 2019-07-21:
Hi
Do You the xdf defination for this ECU,
Turnerpro RT

THANKS
Pete

edwinopie, 2019-07-21:
No i dont, I am not using Tunerpro.

yvan746, 2019-07-22:
I have the same problem with FTDI 232RL vagcom then I will try with Ryan tool and I will see

clevim23, 2019-07-23:
tente a senha FF FF FF FF FF FF FF FF FF FF FF FF FF FF

RyanHope, 2019-07-28:
@edwinopie do you still have access to all the bikes you listed in your op? would it be possible for you to get me the ecmid of each of them so I can update HondaECU

edwinopie, 2019-07-28:
Yes i can. Asap thank yoi..

yvan746, 2019-07-28:
@edwinopie Did you resolve the problem with KKL usb interface?

makunetsumaru, 2019-07-31:
@RyanHope i got same issue as edwinopie.
38770-K45-N42 New CBR 150R K45G (ECM ID: 01 02 04 0f 01)

RyanHope, 2019-07-31:
Need more details about your setup.

makunetsumaru, 2019-07-31:
what kind of detail do you need that can help you?

RyanHope, 2019-07-31:
@makunetsumaru sorry, the details question was a mistake I thought this was a different issue, if you are getting a password failed error then the ecu needs to get backdoord so we can learn what the password is.

edwinopie, 2019-08-11:
Noted

closed 2019-08-11 by edwinopie