MCU-Innovations/HondaECU #40: help CBR500R MJW-T11 Flash result/bad

prostreetprotege, 2019-07-21:
i'm doing XDF file for ecu 38770-MJW-T11 already,but can't flash to ECU
 and than Program notice that>>result/bad<<cloud you help me,please
Ps.Attachments File
[38770-MJW-T11.zip](https://github.com/MCU-Innovations/HondaECU/files/3414785/38770-MJW-T11.zip)

i'm use the TunerPro to edit the xdf file

Best regards

RyanHope, 2019-07-21:
did you enable fix checksum on write?

prostreetprotege, 2019-07-21:
i'm rookie how to enable fix checksum?

prostreetprotege, 2019-07-22:
Now i can flash the bin to my bike,thankyou verymuch

prostreetprotege, 2019-07-22:
Hi ryan
Can you helpme again

1.i'm cannot use read function result/password fail

2.i'm flash on my bike is 20 time
But Now i'm cannot use flash function,data loging,trouble code
Libsusb driver is good
Now i'm not flash new bin on my bike

Plaese Help me
Thankyou so much

Sorry i'm English is a little

RyanHope, 2019-07-22:
I am not following what the issue is. If your flash adapter is displayed in the app then there is no driver issue. If the info panel is reporting something other than OFF/UNKNOWN/OK then your ecu needs recovering, you can do that with a stock bin file from the repo. If the adapter shows up but info panel is showing OFF/UNKNOWN double check your wiring, power supply. If thats all good the kline part of the flash adapter might be fried. That happens often with some of the cheap vag com clones. If the info panel shows OK and you get a password failed error then you ether attempted to change the read password and didnt overwite the bootloader or you wrote a corrupted bin to the ECU. In that case just restore the stock bin.

prostreetprotege, 2019-07-22:
Now on my display show
i'm use lonelec cable to connect
Now i'm cannot flash any bin file
  IMG_20190723_025010.jpg
<https://drive.google.com/file/d/1-BMgK1veXvdRgWE1DSeWeTbFPH__HuRn/view?usp=drivesdk>

  IMG_20190723_011624.jpg
<https://drive.google.com/file/d/17vJohtGFLadDLc43TIQ46HFrhiQquIVn/view?usp=drivesdk>

closed 2019-07-22 by RyanHope