MCU-Innovations/HondaECU #43: Write Problem use 3.0.13

JabaRacing, 2019-07-29:
I tried writing on models 38770-KYJ-971 38770-KVK-xxx 38770-KTW-901 38770-KWT-xxx and all of them during the writing process froze and didn't complete. In the previous version it worked very well.
Has anyone had a problem like this?

RyanHope, 2019-07-30:
I made big changes to the usb/ftdi backend to facilitate a move to a different driver that doesn't conflict with the official FTDI driver. Please pastebin or gist the error log.

prostreetprotege, 2019-07-30:
i have write problem on V3.0.13 and V3.1.0 too

RyanHope, 2019-07-30:
Cant help unless you post your logs and tell me hows its actually failing.

RyanHope, 2019-07-30:
Would also help to know if you are using a diy adapter cable or a commercial one. The faster and more efficient my code gets, the more issues with fake ftdi chips or improper kline circuits become an issue.

JabaRacing, 2019-07-30:
Dear Ryan

I use the DIY adapter suggested, in the previous version worked without
problem.
For better analysis, follows log file collected on a write attempt that
froze by 0.7%

[1.1211] 0 > [fe, 04, 72, 8c]
[1.1336] 0 < [0e, 04, 72, 7c]
[1.1336] 0 > [72, 05, 00, f0, 99]
[1.1451] 0 < [02, 04, 00, fa]
[1.1456] 0 > [72, 05, 71, 00, 18]
[1.1690] 0 < [02, 0f, 71, 00, 01, 00, 8b, 08, 03, 01, 00, 00, 00, 00, e6]
[1.1695] 0 > [72, 05, 00, f0, 99]
[1.1815] 0 < [02, 04, 00, fa]
[1.1815] 0 > [72, 05, 71, 00, 18]
[1.2055] 0 < [02, 0f, 71, 00, 01, 00, 8b, 08, 03, 01, 00, 00, 00, 00, e6]
[1.2055] 0 > [7d, 06, 01, 01, 03, 78]
[1.2294] 0 < [0d, 09, 01, 00, 00, 00, 00, 00, e9]
[1.2294] 0 > [72, 05, 74, 01, 14]
[1.2534] 0 < [02, 0c, 74, 01, 00, 00, 00, 00, 00, 00, 00, 7d]
[1.2534] 0 > [72, 05, 74, 02, 13]
[1.2773] 0 < [02, 0c, 74, 02, 00, 00, 00, 00, 00, 00, 00, 7c]
[1.2773] 0 > [72, 05, 74, 03, 12]
[1.3013] 0 < [02, 0c, 74, 03, 00, 00, 00, 00, 00, 00, 00, 7b]
[1.3013] 0 > [72, 05, 74, 04, 11]
[1.3252] 0 < [02, 0c, 74, 04, 00, 00, 00, 00, 00, 00, 00, 7a]
[1.3252] 0 > [72, 05, 74, 05, 10]
[1.3492] 0 < [02, 0c, 74, 05, 00, 00, 00, 00, 00, 00, 00, 79]
[1.3492] 0 > [72, 05, 74, 06, 0f]
[1.3732] 0 < [02, 0c, 74, 06, 00, 00, 00, 00, 00, 00, 00, 78]
[1.3732] 0 > [72, 05, 74, 07, 0e]
[1.3970] 0 < [02, 0c, 74, 07, 00, 00, 00, 00, 00, 00, 00, 77]
[1.3970] 0 > [72, 05, 74, 08, 0d]
[1.4211] 0 < [02, 0c, 74, 08, 00, 00, 00, 00, 00, 00, 00, 76]
[1.4211] 0 > [72, 05, 74, 09, 0c]
[1.4449] 0 < [02, 0c, 74, 09, 00, 00, 00, 00, 00, 00, 00, 75]
[1.4454] 0 > [72, 05, 74, 0a, 0b]
[1.4689] 0 < [02, 0c, 74, 0a, 00, 00, 00, 00, 00, 00, 00, 74]
[1.4694] 0 > [72, 05, 74, 0b, 0a]
[1.4932] 0 < [02, 0c, 74, 0b, 00, 00, 00, 00, 00, 00, 00, 73]
[1.4932] 0 > [72, 05, 73, 01, 15]
[1.5167] 0 < [02, 0c, 73, 01, 00, 00, 00, 00, 00, 00, 00, 7e]
[1.5172] 0 > [72, 05, 73, 02, 14]
[1.5412] 0 < [02, 0c, 73, 02, 00, 00, 00, 00, 00, 00, 00, 7d]
[1.5412] 0 > [72, 05, 73, 03, 13]
[1.5652] 0 < [02, 0c, 73, 03, 00, 00, 00, 00, 00, 00, 00, 7c]
[1.5652] 0 > [72, 05, 73, 04, 12]
[1.5891] 0 < [02, 0c, 73, 04, 00, 00, 00, 00, 00, 00, 00, 7b]
[1.5891] 0 > [72, 05, 73, 05, 11]
[1.6131] 0 < [02, 0c, 73, 05, 00, 00, 00, 00, 00, 00, 00, 7a]
[1.6131] 0 > [72, 05, 73, 06, 10]
[1.6370] 0 < [02, 0c, 73, 06, 00, 00, 00, 00, 00, 00, 00, 79]
[1.6370] 0 > [72, 05, 73, 07, 0f]
[1.6610] 0 < [02, 0c, 73, 07, 00, 00, 00, 00, 00, 00, 00, 78]
[1.6610] 0 > [72, 05, 73, 08, 0e]
[1.6849] 0 < [02, 0c, 73, 08, 00, 00, 00, 00, 00, 00, 00, 77]
[1.6849] 0 > [72, 05, 73, 09, 0d]
[1.7090] 0 < [02, 0c, 73, 09, 00, 00, 00, 00, 00, 00, 00, 76]
[1.7090] 0 > [72, 05, 73, 0a, 0c]
[1.7330] 0 < [02, 0c, 73, 0a, 00, 00, 00, 00, 00, 00, 00, 75]
[1.7330] 0 > [72, 05, 73, 0b, 0b]
[1.7569] 0 < [02, 0c, 73, 0b, 00, 00, 00, 00, 00, 00, 00, 74]
[1.7574] 0 > [72, 05, 71, 10, 08]
[1.7689] 0 < [02, 05, 71, 10, 78]
[1.7689] 0 > [72, 05, 71, 11, 07]
[1.8049] 0 < [02, 19, 71, 11, 00, 00, 21, 06, 9d, 4d, 9c, 3f, 93, 64, ff, ff, 75, 00, 00, 00, 80, 5f, 36, 2e, ca]
[1.8049] 0 > [72, 05, 71, 13, 05]
[1.8288] 0 < [02, 05, 71, 13, 75]
[1.8288] 0 > [72, 05, 71, 17, 01]
[... 1820 lines cut ...]
[12.8761] 0 < [02, 04, 00, fa]
[12.8761] 0 > [72, 05, 00, f0, 99]
[12.8881] 0 < [02, 04, 00, fa]
[12.8881] 0 > [72, 05, 00, f0, 99]
[12.9000] 0 < [02, 04, 00, fa]
[12.9000] 0 > [72, 05, 00, f0, 99]
[12.9120] 0 < [02, 04, 00, fa]
[12.9120] 0 > [72, 05, 00, f0, 99]
[12.9240] 0 < [02, 04, 00, fa]
[12.9240] 0 > [72, 05, 00, f0, 99]
[12.9360] 0 < [02, 04, 00, fa]
[12.9360] 0 > [72, 05, 00, f0, 99]
[12.9480] 0 < [02, 04, 00, fa]
[12.9480] 0 > [72, 05, 00, f0, 99]
[12.9599] 0 < [02, 04, 00, fa]
[12.9599] 0 > [72, 05, 00, f0, 99]
[12.9719] 0 < [02, 04, 00, fa]
[12.9724] 0 > [72, 05, 00, f0, 99]
[12.9839] 0 < [02, 04, 00, fa]
[12.9839] 0 > [72, 05, 00, f0, 99]
[12.9959] 0 < [02, 04, 00, fa]
[12.9959] 0 > [72, 05, 00, f0, 99]
[13.0078] 0 < [02, 04, 00, fa]
[13.0083] 0 > [72, 05, 00, f0, 99]
[13.0198] 0 < [02, 0

RyanHope, 2019-07-30:
Did you build the adapter exactly like it was shown in the diagram or did you substitute resistors like I claimed was possible in the readme? Have you seen the comments in this thread (https://github.com/MCU-Innovations/HondaECU/issues/37). I too started getting lots when I changed the usb library that I was using, and fixing my adapter solved all my issues. If you made your adapter with 1k resistor on the kline wire on the bike side, try replacing it with a 1.2k or add a 200ohm inline.

JabaRacing, 2019-07-30:
I built exactly with the values in the diagram, 1.2k
Before reporting the error, I read topic 37.
Anyway I'll try the other resistor values to see if it works.
Just as a comment, when in Data Logging mode the communication is working
properly without "freezing".
thank you

RyanHope, 2019-07-31:
@JabaRacing I just noticed that I had some uncomitted changes in eculib (which HondaECU depends on), so, when testing on my machine things work different than they should in the built 3.1.0 builds. I updated things to fix this, please try out 3.1.1 and let me know if you still have issues.

JabaRacing, 2019-08-02:
Hello Ryan

I did some testing with version 3.1.1, read, write, data logging ... it's
working very well. Congratulations !
This new version does not generate errors when during data logging you turn
the ECU on and off. it is much better.

RyanHope, 2019-08-02:
I still need to fix some of the displays and remove old data when power is toggled. Glad is working though.

closed 2019-08-02 by RyanHope