MCU-Innovations/HondaECU #44: Honda Pcx125 year 2010 read and flash

alfo1974, 2019-08-02:
Data logging and trouble codes work very well on my scooter, with OBD2 Galletto 1260 usb and libusbK driver, of course.
Ecu number is: 30400-KWN-643. I'd like to read and flash it, but after "turn off and on" the ignition, nothing happens.
[debug pcx125.txt](https://github.com/MCU-Innovations/HondaECU/files/3460996/debug.pcx125.txt)

RyanHope, 2019-08-05:
have you tested on 3.1.2?

alfo1974, 2019-08-05:
Even with 3.1.2 and some test, after "Turn on Ecu", Ecm ID: 01 00 ca 01 02 goes from green to grey.
[Pcx 2 debug.txt](https://github.com/MCU-Innovations/HondaECU/files/3467505/Pcx.2.debug.txt)
[image]

alfo1974, 2019-08-06:
Ryan, same situation with 3.1.3. "Read" doesn't start. Anyway I'd like to thank you for your great job!

RyanHope, 2019-08-08:
Ok here is the deal, I just tagged a v3.1.5 release. In the file menu is now an option to bring up a settings dialog where you can change 2 settings. The number of retries after a failed command, and the timeout for waiting for a response. If you are having issues with reads or write stopping suddenly, try turning up the retries first. When I had it at 10 there were no complaints about things not working. Let me know if this helps or not.

alfo1974, 2019-08-09:
From debug file it seems to me that something has changed (10 retries and 1 second timeout), but reading still doesn't start. 
[pcx debug3.txt](https://github.com/MCU-Innovations/HondaECU/files/3485872/pcx.debug3.txt)

RyanHope, 2019-08-09:
So the main issue is the on/off detection?

alfo1974, 2019-08-09:
Exactly. After "Preparing to read ecu.. Turn off and Turn on", the software talks to the ecu (ECM ID remains green), but isn't able to read...

RyanHope, 2019-08-09:
Ok this on/off detection has been the biggest pain to solve cause every adapter behaves slightly different. I have actually implemented a few different techniques for detecting the power and now that I have a settings dialog I could expose them so people can chose the one that works most reliably for their adapter.

alfo1974, 2019-08-09:
I understand. I'll make some other tests with different adapters and different settings, then. Your software is great!

RyanHope, 2019-08-09:
Alright so 3.1.7 is building now, should be in the releases by the time you see it. In the settings menu there is now a combo box to select from 2 different kline detection methods. "poll_modem_status" is the approach that I think is most correct and clean... non-hackish. "poll_modem_status" should work flawlessly with the optocoupler circuit that I have in the readme. This is the method I used from the very first version of this project. The second method "loopback_ping" is a more hackish method that I introduced when people started using the vag com clone cables. The "poll_modem_status" method does not work with the vag com cables and as a result the "loopback_ping" method became default a while ago. Everyone should use "poll_modem_status" if possible, if "loopback_ping" doesn't work there are a few parameters in that function that I can expose. Otherwise I have to do a little more digging into the issues.

RyanHope, 2019-08-09:
3.1.7 had a silent bug, fixed with 3.1.8

closed 2019-08-18 by RyanHope

alfo1974, 2019-08-19:
Not even HondaECU-v3.3.0 works. Same on/off detection problem, even with different settings.