MCU-Innovations/HondaECU #45: ECT sensor : read error

JabaRacing, 2019-08-02:
Hello

Hello
In a bench test, I always see an ECT sensor read error. The value shown is 14C higher than the actual temperature.
Have other users noticed this error?
Below is a print showing the correct value IAT and the highest value 14C ECT.
[image]

alfo1974, 2019-08-02:
ECT (Engine Coolant Temperature) and IAT (Intake Air Temperature) are different sensor. Maybe your engine is still warm.

RyanHope, 2019-08-02:
The model error is fixed, get a live build from the autobuilder or wait
till I tag a new release shortly.

JabaRacing, 2019-08-02:
Thanks Ryan

JabaRacing, 2019-08-02:
Not a test on the engine, but a bench test. The 2 sensors are the same
temperature.

RyanHope, 2019-08-02:
What table number is your main data table, 0x10, 0x11, 0x17 something else?

JabaRacing, 2019-08-02:
The table number is 0x11

RyanHope, 2019-08-02:
Just tagged 3.1.2 with a few small fixes, let me know if that improves anything.

JabaRacing, 2019-08-02:
Dear Ryan

More tests in 3.1.2, functions ok ( write, read, etc ) but the same ECT
error

[image: image.png]

RyanHope, 2019-08-18:
Still having issues?

JabaRacing, 2019-08-19:
Hello Ryan

The same issue at v3.3.0
[image: image.png]

RyanHope, 2019-08-19:
What exactly is the error, I cant tell from the photo. I dont see any error related the ECT...

JabaRacing, 2019-08-19:
Ryan
In this photo the two sensors (IAT ECT) are at the same physical
temperature 22C. But the reported values are 14 degrees apart.
This is the read error I've been encountering for the 38770-KVK-xxx
38770-KWT-xxx ECUs

[image: image.png]

thanks

RyanHope, 2019-08-19:
Is this with the ecu actually in the bike or on the bench?

JabaRacing, 2019-08-19:
This is with the ECU on the bench using the bike's original sensors.
But when the bike has been stationary for a long time (one week) the value
has the same error.