MCU-Innovations/HondaECU #46: European ECU and HISS

ninjarmoni, 2019-08-02:
Hi Ryan, 

2 questions for you, didn't how to reach you...

I'm planning to order this cable : https://www.lonelec.co.uk/Honda-Motorbike-ECU-Flasher-Cable 
First question is do you know if it works with the last version of HondaECU ?

I want to replace my french CBR 1000 RR 2008 ECU map (38770-MFL-622) by loading the 38770-MFL-644. Second question : what about HISS, can I keep it or do I have to delete this function ?

Thanks,
Romain

RyanHope, 2019-08-03:
I can send you a bin with immo disabled.

ninjarmoni, 2019-08-03:
Hi Ryan, 

In France, already in 2008 and until 2016 we had a restriction law about motorcycle horse power. The maximum was 106 hp (approx. 78 Kw). So for my '08 CBR1000RR in order to be "full power" I ordered a used ECU on eBay (model 38770-MFL-D21) to replace the genuine one (model 38770-MFL-622).

To do so, can I use your software to flash the 38770-MFL-D21 I bought ?
And then with this bin you mentioned earlier with immo disabled, it won't ask any special key to start ?

Thanks for your time and help.

Cheers,
Romain

RyanHope, 2019-08-03:
You could have flashed the unrestricted EU bin over your French restricted
ECU...

ninjarmoni, 2019-08-04:
Yes, but I want to be able to go back in case of selling the bike because according to what I understand the key management (H.I.S.S.) will be lost, right ?

RyanHope, 2019-08-04:
Disabling the immo is one bit in the flash rom, can't be turned back on at
any point. The HISS key info is stored in the eeprom which does not get
erased when you flash the rom.

ninjarmoni, 2019-08-04:
Okay, so if I flash another ECU (from eBay) my bike won’t start with my actual keys ?

RyanHope, 2019-08-04:
Not unless you disable the immo or copy over the eeprom

ninjarmoni, 2019-08-04:
With HondaECU can we do that ?

RyanHope, 2019-08-04:
With HondaEcu you can flash a bin with immo disabled. I haven't added any
eeprom tools to HondaECU yet

ninjarmoni, 2019-08-04:
Alright, immo disabled means no more HISS ?
Sorry for my newbie’s questions !

RyanHope, 2019-08-04:
Yes, you immo=immobilizer=HISS

ninjarmoni, 2019-08-04:
As soon as I have received this cable (https://www.lonelec.co.uk/Honda-Motorbike-ECU-Flasher-Cable) I’ll try what you advised me : flash the unrestricted EU bin over my French restricted
Thanks!

closed 2019-08-05 by RyanHope

ninjarmoni, 2019-08-09:
Hi Ryan, 

I've tried to flash the ECU and it finally worked after several crashes "write interrupted". Even if the flash is OK now I can't start the bike due to HISS. 

As you proposed earlier, can you send a Bin with IMMO disabled ? 

CBR 1000 RR 2008 (38770-MFL-622 was at first, now it's 38770-MFL-671)

Thanks !

ninjarmoni, 2019-08-10:
It finally worked with 38770-MFL-D21 with HISS activated, that's awesome ! Great Job Man 👍