MCU-Innovations/HondaECU #47: CBR300 ecu, Eculib file with pycharm doesnt found.

Josileissu, 2019-08-04:
Hi,

I'm using PyCharm to test this code for remap my motorcycle. Can you help me step by step to use your code?

I have access to cbr's on my brothers office and i wanna help he to adjust injection...

Att,

RyanHope, 2019-08-04:
Eculib is in another one of my repos so that I can use the code in other
projects.

Josileissu, 2019-08-04:
Oh, sorry.

I see that exist an exe file. So what i need to remap my motorcicle (cause im adjusting my cbf 250 with pgmfi from cb300r), is use your HondaEcu.exe to extract bin file from ecu? After this i use turnerpro to do modifications and write again use HondaEcu in the ECU ?

Sorry for my english, i'm lazzy to use translate ¬¬ haha.

thx

RyanHope, 2019-08-04:
You will need an xdf for the cb300 before you can do anything in TunerPro
but you have the general process down.

Josileissu, 2019-08-04:
Sorry for the inconvenience.

1° - Can i obtain an xdf file from the motorcycle?
2° - How to do this?
3° - Where is the general process?
4° - The wiring diagram that we need to do, the 4N25 optocouplers, resistors and led... U has an image from this device?
5° - Do you know the pgmfi works flex fuel, i will take an problem with this using tunerpro?

I'm not a eletrical engineering, and not a technicall, but i understand quickly if u help me :)

Thx

Josileissu, 2019-08-04:
FTDI is this device? That i need to to this wiring adjust for use to receive the files?
[image]

Wiring adjust.
[image]

Pc serial port is that sector in the begin on pcb?

Josileissu, 2019-08-04:
or is this?
[image]

RyanHope, 2019-08-04:
Both will probably work

Josileissu, 2019-08-04:
Can u send an image that device you use? For i see? Then i copy you :)

Josileissu, 2019-08-04:
I'm searching in the net, i found this device.
So you use that i show first.
[image]

https://gonzos.net/projects/ctx-obd/

closed 2019-08-05 by RyanHope