MCU-Innovations/HondaECU #49: Read FTDI

Josileissu, 2019-08-06:
Hi,

How can i read the ftdi conversor direct from the pc to obd port?

I try use TTL but i cant do this. 

OBD Port - CB300
[image]

I'm using protoboard to test the ryan circuit to separate obd from ftdi
[image]

[image]

I bought the ftdi with ft232 but i cant read this from my computer. How can i do this? What i need?
Need usb 2.0? I cant plug into a 3.0?

Josileissu, 2019-08-06:
i've tried to use ttl with hondaecu.exe but i know this dont work cause the .exe needs ftdi.

TrustyTristan, 2019-08-06:
Your breadboard looks wrong. The Optocoupler's look like they are on the same connected to the same tracks.

See this
[image]

Josileissu, 2019-08-06:
Oh, thanks. My brother helps me to connect on to breadboard.

I've try to connect FT232RL on my computer and i got it. Just use the correct usb cable will appear in com serial. After installing libusbk they will appear in a new item in device manager. And after this, is ok to connect with HondaEcu.exe

So the next step i need to do is to connect with my cb300 to get parameters. haha' Discovering a new world :O

Josileissu, 2019-08-07:
Here is the breadboard mounted, thx for helping. I show for my another brother (hahaha, here we ate in 7 brothers), and he helps me to adjust.

[image]
[image]
[image]
[image]
[image]
[image]

Josileissu, 2019-08-07:
Here, i cant read. Wat im doing wrong?

[image]
[image]
[image]

Josileissu, 2019-08-07:
Ok, i ger an unknow bike
[image]
[image]

Josileissu, 2019-08-07:
I created an file call cb300.bin without text.
And put in the file field.
Is this method correct?

[image]

RyanHope, 2019-08-07:
What is the ECU part number and other model info and I can update the code

Josileissu, 2019-08-07:
The model of motorcycle is: CB300R 2015
Keihin ECU: 38770-KVK-B23
Flex fuel

Josileissu, 2019-08-07:
here, i take a picture from the ecu. My brother almost burst the connector hahaha. Join the 12v with gnd oO"
[image]

RyanHope, 2019-08-08:
Thanks I will add this info to the next release.

Josileissu, 2019-08-08:
I thank you that. Is this release are include this ecu, 3.1.5?

prostreetprotege, 2019-08-08:
Hey Ryan. Can you add my ecmid on next release

38770-MJW-T11

Thankyou so much

vasiliouv84, 2019-08-08:
Hello Ryan can you add ecmid 38770 mee 643?
Thanks in advance

Josileissu, 2019-08-08:
Hi, i've trying to use in pycharm your code to test... And i have an doubt, what will happen after i connect the diagnostic port from motorcycle to my pc using you software?

Regards.

[image]

clevim23, 2019-08-09:
[image]

clevim23, 2019-08-09:
Hello I have this error when trying to read the ECU XRE 300 12

clevim23, 2019-08-09:
[log.txt](https://github.com/MCU-Innovations/HondaECU/files/3488199/log.txt)

Josileissu, 2019-08-10:
Hi,

I connected with the last release but the log file doesnt stop
[image]

What i'm doing wrong?
[logcb300.txt](https://github.com/MCU-Innovations/HondaECU/files/3488427/logcb300.txt)

RyanHope, 2019-08-10:
Doesn't look like anything is wrong.

Josileissu, 2019-08-10:
What i need to do? I just put the ftdi on the connectors and wait? Doesnt need insert bin file in the field for this?

Josileissu, 2019-08-10:
Here i tested running, but only o2 sensor goes.

Josileissu, 2019-08-10:
[image]

edwinopie, 2019-08-11:
Hi ryan,
Can you add 38770-K84-901 on next Hondaecu release? I got unknown model too..

RyanHope, 2019-08-11:
What is the ecmid and model info?

edwinopie, 2019-08-11:
Model : CRF150L 
ECM ID: 01 02 32 0f 01

RyanHope, 2019-08-12:
@Josileissu  what do you mean "only o2 sensor goes"?

RyanHope, 2019-08-12:
@edwinopie added the info with commit 22e423a4d4fa812216f2e0bf7c2d5c89e210d34f, can you send me the bin for this bike as well so I can update the collection?

closed 2019-08-12 by RyanHope

edwinopie, 2019-08-12:
I can. I was using other flasher, But i am not sure it was the right reading address.. or let.say the right bin size file..

RyanHope, 2019-08-12:
If you can email me all the read attempts you have done I can check things
over.

edwinopie, 2019-08-12:
done, emailed to you