MCU-Innovations/HondaECU #53: ECMIDs and Model Info

RyanHope, 2019-08-12:
Use this thread to request updates to the ecmid database with model specific meta data. For each bike I need the ECMID, ECU part number and the model name and year.

edwinopie, 2019-08-12:
this thread that i have been waiting for.. thank you..

JabaRacing, 2019-08-12:
Thanks Ryan !

ECMID: 01 03 16 08 01
 ECU part number : 38770-K99-B02
 Model name : CRF250F
 year : 2019

clevim23, 2019-08-12:
> Obrigado Ryan! ECMID: 01 03 16 08 01 Número de peça ECU: 38770-K99-B02 Nome do modelo: CRF250F ano: 2019 Em seg, 12 de ago de 2019 às 11:16, edwinopie <notifications@github.com> escreveu:
> […](#)

Could you read and write the ECF CRF 250F?

clevim23, 2019-08-12:
ECU

JabaRacing, 2019-08-12:
I'm working on it

edwinopie, 2019-08-12:
ECM ID: 01 02 32 0f 01
ECU PartNumber : 38779-K84-901
Model : CRF150L
Year : 2017

SparkPuck, 2019-08-13:
Hi Ryan,

thank you for the great job! I've successfully read the binary of my CB600F. My vag-com 409 cable is working out of the box. I probably had luck with the ftdi chip.
The bin is attached.
[38770-MGM-D11.zip](https://github.com/MCU-Innovations/HondaECU/files/3496585/38770-MGM-D11.zip)

Model: CB600F ABS Hornet PC41
Year: 2007 - 2012
Region: Europe
ECMID: \x01\x00\xe0\x01\x01
ECMIDAddr: 0x2290F
PN: 38770-MGM-D11

Now i'm trying to find the address of the flag to enable/disable deceleration fuel cutoff. In the cbr500r bin/xdf it's two bytes before the ECMID. May someone can help me?

Regards

RyanHope, 2019-08-13:
@SparkPuck I already have this is ecmid in my db as a CBR600F, I am not sure how people want situations like this displayed. We have this problem other size CB/CBR's that share the same ECU. What do people like better:

1) CB/CBR600F
2) CB/CBR 600 F
3) CB600F, CBR600F

I am thinking of normalizing everything to be like 2.

SparkPuck, 2019-08-14:
@RyanHope I also prefer no. 2. But no. 3 also would be fine. I think it would be very usefull if we can include the Engine Code into the Modell description. (for Example PC41 in this case or PC40 for the CBR600RR Model) The Engines can be very different. The A in "FA" somtimes added to tell apart the abs version.

1. CB/CBR 600 F/FA PC41
2. CB/CBR 600 F PC41