MCU-Innovations/HondaECU #54: Writing: Interupted - ECU Broken

devimauz, 2019-08-12:
Hey
Read, DTC, Live Data worked.
Tired to reflash it and after erased Writing:Interupted error came and ECU Light off.
With killswitch off some data coming but... what to do?
When I turn killswitch ON try to Erase the ECU again but its fails.
Here is the log.
[fail.txt](https://github.com/MCU-Innovations/HondaECU/files/3493584/fail.txt)

Thanks!

anctech, 2019-08-13:
Hi , you need write use V3.0.12 working perfect

devimauz, 2019-08-13:
Need to recovery first?

anctech, 2019-08-13:
Yes

devimauz, 2019-08-13:
But mine have M8C (r5f2134), Should be M16C, No?

anctech, 2019-08-13:
No , R5F2134 is M16C with 48kb size bin

devimauz, 2019-08-13:
How?
[image]
 For this chip shows 24k size... What should I choose then?

anctech, 2019-08-13:
Read use HondaEcu by Kline working perfect !

anctech, 2019-08-13:
For recovery only need write again use original file

Pada tanggal Sel, 13 Agt 2019 20:55, ECU REMAP / REFLASH INDONESIA <
anctech.indonesia@gmail.com> menulis:

devimauz, 2019-08-13:
What CPU should I choose from the list?

devimauz, 2019-08-13:
I will make a recovery tutorial if I finished with this.
[image]

anctech, 2019-08-13:
I cannot advice you , because you not understand .

Your ecu brick ?
cannot read engine data , dtc and other .
only can flash ?

you know your ecu need RECOVERY !
how to recovery , you only need write again bin file original to your ecu .
use Ryan Software HondaEcu .
no need cut ecu !!!!
no need FDT software

Understand ?

anctech, 2019-08-13:
[image]

for RECOVERY , WRITE use this SOFTWARE

devimauz, 2019-08-13:
How? Flash is gray for me!

anctech, 2019-08-13:
PLUG the wiring to ecu please .
12v
gnd
kline

and put power 12v dont forget , and i want to see the software , take picture for me

devimauz, 2019-08-13:
[image]

devimauz, 2019-08-13:
[image]

anctech, 2019-08-13:
12v or Switch have on ?

devimauz, 2019-08-13:
Switch on

anctech, 2019-08-13:
R5F21347H

ok , try this

devimauz, 2019-08-13:
Is this ok?
[image]

RyanHope, 2019-08-13:
Are you writing that to the correct location?

devimauz, 2019-08-13:
It don't want to write more.
[image]

RyanHope, 2019-08-13:
Just write to the entire user flash starting at 0x4000

devimauz, 2019-08-13:
How?

1. I connect to device
2. Download Active File

And it says file exceedds the flash ROM size.

anctech, 2019-08-13:
your ecu can't be saved anymore.  same as mine

devimauz, 2019-08-13:
Uhm, Why? Flashing over PCB should work.

RyanHope, 2019-08-13:
The ECU can always be recovered with Renesas FTD, just select the entire
flash area and start writing the stock bin from github starting at 0x4000.

devimauz, 2019-08-13:
But at the selection I can upload and delete. Can you show picture for us
where to choose?

Ryan Hope <notifications@github.com> ezt írta (időpont: 2019. aug. 13., K
18:05):

devimauz, 2019-08-13:
Looks like the FDT start reading the file from 04000h and I have to FILL with 00 or FF until that address.

devimauz, 2019-08-13:
Okey, Data part deleted too. We need data from 0x00 to 0x4000

devimauz, 2019-08-13:
Please copy these files to 
"c:\Program Files (x86)\Renesas\FDT4.09\kernels\ProtD\R5F21347H\Renesas\1_0_00\"
[R5F2134H_Modified.zip](https://github.com/MCU-Innovations/HondaECU/files/3498423/R5F2134H_Modified.zip)
and read it all and upload to here please.
[image]
Maybe this is the way to reprogram a broken ecu.
0x3000-0x4000 looks like something data part.
Thanks!

RyanHope, 2019-08-13:
This is the Monkey ecu in FDT, it has the user and data sections properly defined.
[image]
I have never messed with the data portion...

RyanHope, 2019-08-13:
I also dont use the Tree tab, I use the Address tab

devimauz, 2019-08-13:
But Address tab doesn't allow to read under 4000. I messed with it.(Accidentally) So If available please read grom ecu for me. This called full bdm read. Then I can recovery my ecu. Thanks!

RyanHope, 2019-08-13:
I guess I am confused why you needed to read under 4000? Erasing the user flash should have only touched 0x4000 and up and then writing back at 0x4000 would leave the data section untouched. I think the issue is that in the image you posted the user flash starts at 0x0 which is wrong. I didnt get a chance to look at the changes in the files you posted.

devimauz, 2019-08-13:
Because I tired to dissemble it and accidentally i not saved it so I destroyed. Usually these have iRom inside that you can't read like this. So.

devimauz, 2019-08-13:
Do you have any Grom ECU that you can read?

RyanHope, 2019-08-13:
Yeah I have 2 softbricked K26-C31 ecus that I need to recover for someone
and I have 2 K26-911, one in my OG grom and one as a backup.

devimauz, 2019-08-13:
Can you read the 911 from 3000 to 4000, please?

devimauz, 2019-08-14:
> […](#)

Why cant save yours?

RyanHope, 2019-08-14:
@devimauz  yeah I will try and do a read today

devimauz, 2019-08-14:
Thank you man! Next week I get backup ecu.

Ryan Hope <notifications@github.com> ezt írta (időpont: 2019. aug. 14., Sze
16:04):

devimauz, 2019-08-14:
@RyanHope don't forget me please! :(

devimauz, 2019-08-15:
@RyanHope ? :(

edwinopie, 2019-08-20:
Hi. If i am going to use FDT to recover, how to connect to ecu (33pin)? Should it be a direct connection? Or i can use trough bike DLC while the ecu connectes to bike wiring?

RyanHope, 2019-08-20:
It should not be connected to the bike or external power in any way. All
the connections are from the FTDI breakout board.

edwinopie, 2019-08-20:
Direct to microprocessor? Ieu dismantle the ecu?

devimauz, 2019-08-20:
@RyanHope Please can you read it for me?

devimauz, 2019-08-22:
Fixed with another ecu.

closed 2019-08-22 by devimauz