MCU-Innovations/HondaECU #55: Hondaecu 320 tes read CRF150L

edwinopie, 2019-08-13:
hi
try to read, i think its reading. i got 48kb bin file. but progress reading is not seen(the green bar)
[tes read flash crf150l 4000.txt](https://github.com/MCU-Innovations/HondaECU/files/3495437/tes.read.flash.crf150l.4000.txt)

[tes crf read hondaecu 4000.zip](https://github.com/MCU-Innovations/HondaECU/files/3495440/tes.crf.read.hondaecu.4000.zip)
start offset 4000

RyanHope, 2019-08-13:
Try reading from 0x7000

edwinopie, 2019-08-13:
ok. how is the bin size should be? more than 48kb?

RyanHope, 2019-08-13:
im not sure on this ecu, I assume you tried reading from 0 and that failed?

edwinopie, 2019-08-13:
Reading complete. But result is bad.. that written on small pop window.. i
forget to screenshot. Try again soon.. my kkl ft232rl is burning i guess..
i dont know why

edwinopie, 2019-08-14:
Hi. Can i use any tipe FTDI chip for this? Or there ia a specific type of th FTDI?

closed 2019-08-18 by edwinopie