MCU-Innovations/HondaECU #56: Which USB FTDI Cable working good with Hondaecu?

edwinopie, 2019-08-15:
hi,
Kindly to share which cable working well with HondaECU?

RyanHope, 2019-08-15:
I have not found a commercial cable that I like yet. I have tried the cheap VAG COM KKL clones on ebay, they work for a while then eventually burn out. I've had the best luck building the circuit that is on my github page. I am working a new solution that should be really cheap and available for the world wide market.

edwinopie, 2019-08-15:
thats explain why my vag com kkl burn out too...

edwinopie, 2019-08-15:
any specific FTDI type for hondaECU?

RyanHope, 2019-08-15:
Real FTDI chips are more robust to poor signals than the clone chips are, but if the kline conversion part of the circuit is correct everything should work. In addition to the opamp circuit I also made a circuit with transistors that worked well. The opamps are very robust to electrical spikes and wont burn out like the opamps used in the vag com cables. I do however worry that the opamps are too slow and are causing issues with higher baud rates.

edwinopie, 2019-08-15:
i found USB to Kline interface from europe, plan to order it and give it a try.. it is using original FTDI chip.

edwinopie, 2019-08-18:
what is the protocol of the ecu? is it K-Line (ISO 9141)?

RyanHope, 2019-08-18:
the protocol is proprietary

edwinopie, 2019-08-18:
can i use this cable which has FTDI RS232 to USB adapter with an ST L9637 line driver?

RyanHope, 2019-08-18:
if its a kline adapter with an ftdi chip it should work (tm)... thats all I can say

edwinopie, 2019-08-18:
noted, thank you

closed 2019-08-18 by edwinopie