MCU-Innovations/HondaECU #57: Read buttons greyed out

uszzz, 2019-08-16:
Hi,

Quick question: what is a proper procedure of reading flash or EEPROM? I get a connection with ECU (state lamp green, identification is OK) but reading is not possible (read button greyed out in Reading mode), either I enter a file name or not. 

What am I doing wrong?

Regards;
uszzz
[image]

RyanHope, 2019-08-16:
Click the security access button next to the device list to enable reading
the flash and eerom features, need to power cycle ecu to get back to data
logging and write flash

uszzz, 2019-08-16:
Thanks for advice Ryan. Unfortunatelly it gets stuck on Turn On ECU. I can see HE gets connection with ECU again but the popup Turn On ECU does not disappear and reading is not possible. Tried 2 ECUs, both supported by HE. 

Regards;
uszzz

RyanHope, 2019-08-16:
I believe I fixed that issue but I haven't tagged a new release yet. Try this devel build: https://ci.appveyor.com/api/buildjobs/urkkhpykffbaoons/artifacts/dist%2FHondaECU-v3.2.1-1-g5bbbfcf.exe

uszzz, 2019-08-16:
Sorry to say that but the devel build works the same way - it stucks on Turn On ECU pop-up.

RyanHope, 2019-08-16:
I got similar feedback from someone else as well. I am adding a new setting
pertaining to the on/off detection that I am hoping might help. What
bike/ecu are you having issues with again? Is this with ecu on bike or on
the bench?

uszzz, 2019-08-16:
The bike is CBR600RR 38770-MFJ-F03 ECU. ECU is on the bench. Also tried (not devel build) on CBR1000RR 2009 ECU with same result.

RyanHope, 2019-08-16:
Alright I have tagged a new version to test 3.2.2. In this version there are some new options in the settings menu that might help people having issues. However, I changed all the defaults to try and work for more people... end up slowing things down a little. So if it does work with the new settings try dialing back retries and timeouts a little. The new setting really worth playing with is the number kline test bytes, new default is 2, used to be 1, maybe more will help, not sure.

uszzz, 2019-08-16:
Ryan, I cant find settings that would work for me. Tried to play with all the settings and some combinations. I am using Lonelec cable which works flawlessly on all k-line based ECUs. More - I tried to read the CBR600 ECU with Honda ECU 3.0.2 and it works great at 375bps. So I suppose the setting are/were there. I wonder why the application shows a correct communication with ECU (after power cycle) but Turn On ECU pop-up is still there...?

RyanHope, 2019-08-16:
I switched ftdi backend libraries recently and it seems less tolerant of
low quality chips and circuits, not sure why, still looking into it. The
latest on the autobuilder
https://ci.appveyor.com/api/buildjobs/sd6t94xn98pyv2fc/artifacts/dist%2FHondaECU-v3.2.2-2-gd3a15a2.exe
has all exception handling turned off so give it a test and see if you can
get any weird errors to come up. After the switch to this new backend it
highlighted a resistor issue I had in my circuit and after fixing that I
never had an issue again so its extra hard to debug.

RyanHope, 2019-08-16:
I believe I have it all straightened out in v3.2.2-3:

https://ci.appveyor.com/api/buildjobs/vi0e0lk4s12r5pqu/artifacts/dist%2FHondaECU-v3.2.2-3-gca98ecc.exe

clevim23, 2019-08-17:
I also have problems reading and writing with versions 13.2 up

RyanHope, 2019-08-17:
I have figured out the write interrupted errors, not sure whats up with the on/off detection for some adapters yet. @clevim23  are your reads failings or just having issues getting into security access mode?

clevim23, 2019-08-17:
successfully enters safe mode, but read and write options are grayed out

clevim23, 2019-08-17:
3.0 versions work perfectly

RyanHope, 2019-08-17:
to write flash you need ECUSTATE.OK (green led bottom left corner), for read flash and to read/write eeprom you need to be in security access mode (purple led bottom left corner)

clevim23, 2019-08-17:
Yes green dot OK after running purple dot safe mode OK, after entering name of or file to be read or written the commands turn gray

uszzz, 2019-08-17:
Still no luck for me. Neither version 3.2.2-2 nor 3.2.2-3 work as expected.

RyanHope, 2019-08-17:
Yeah im still working on the last touches to another release, these issues
only happen in windows (not linux). I can reproduce a lot of issues now so
hang tight.

uszzz, 2019-08-17:
I never had an issue on any k-lined ECU when using Lonelec cable on any Windows version (XP/7/8/10). The only difference is driver - I've been using FDTI drivers. As far as I understand libusbK is the only option for you?

RyanHope, 2019-08-17:
This has nothing to do with libusbk...

uszzz, 2019-08-17:
I see. Just wanted to confirm Lonelec is a very stable cable.

RyanHope, 2019-08-17:
This isnt a cable issue either... its just a windows/usb issue.
Different libraries handle things differently and when I switched
libraries I didn't make all the necessary changes. I'm pretty close to
having something that seems stable on Windows now.

uszzz, 2019-08-17:
Good to hear that. 
3.2.3 still does not work.

uszzz, 2019-08-17:
Unfortunatelly 3.2.4 makes no change to me.

RyanHope, 2019-08-17:
For the on/off issue or incomplete writes?

uszzz, 2019-08-17:
On/off issue. I didnt test writing yet.

RyanHope, 2019-08-17:
Are you getting any new popup windows with errors or is it just silent?

uszzz, 2019-08-17:
Silent. Turn On ECU stays on.
One of last devel builds was giving some error windows. Can test it again if you need the output.

RyanHope, 2019-08-17:
Most of the changes I have been making in the past day should address the
incomplete read/writes on windows. Haven't specifically looked at the
on/off stuff. Let me know if you can write on this build:
https://ci.appveyor.com/api/buildjobs/x7jiwy8eg0xvotp3/artifacts/dist%2FHondaECU-v3.2.4-4-gd20b529.exe

uszzz, 2019-08-18:
With v. 3.2.5 I can read flash and EEPROM but software does not recognize ECU model/year in Security mode (don't know if it is intended). After a power cycle model/year is recognized properely and I can switch to write mode. Writing has completed succesfully. 
Thanks.

RyanHope, 2019-08-18:
There are no commands to send while in security access mode that give an identifiable ID, so, I either have to cache the model info before the power cycle or display something generic. If you close the app and then start it back up and the ecu is in security access mode I wont have any cached info to go by.

closed 2019-08-18 by RyanHope