MCU-Innovations/HondaECU #59: Address to Start/Finish read

edwinopie, 2019-08-18:
wondering filling address as a start mark to read flash, is the honda ecu knows the address by itself where to finish read? because its related to the size of the bin file, or it detect atuomatically?

RyanHope, 2019-08-18:
HondaECU stops reading when it stops receiving data in the responses. The only critical value seems to be the starting location.

edwinopie, 2019-08-18:
noted, 

for an example :
CRF110F | 2019 | 64 kb | 38770-KYK-D12 | 0 | 8000 
the file size is 64kb, start read from 8000, it should be end read at 17FFF, right?

so if unknown ecu, unkown bin size? how to find out those info(bin size, start read address, finish read address)

RyanHope, 2019-08-18:
There is no way to know 100% in advance. You can make some assumptions based on a comparison of its physical size and part number to other similar ECUs. You can just try reading at different offsets until you get a good read. You can cut into one and get the chip part number then lookup the memory layout in a data sheet.

edwinopie, 2019-08-18:
thanks for the information, give me another view about reading scheme of HondaECU. 
On assumption side as you said, you mean similar part number, like 38770-xxx-901 is maybe identicall to 38770-yyy-901, can i assume that way?

RyanHope, 2019-08-18:
No its the middle letters that matter. 38770 is Hondas part code for
ECU...the middle 3 letters identifies the ECU hardware and the last 3
is the region/software differences. With very few exceptions if the
middle 3 letters match, then the ECU is bin compat with other software
versions. That doesn't mean the bike will run as some regions are
missing sensors or have immobilizers, etc. But there also seem to be
other patterns, all the 38770-K??-??? ecus seem to be based on m16c
MCUS, the CB/CBR 600/1000's use m32r, the CRFs and newer mid sized
bikes (cbr500, cb650) all seem to use PPCs.

edwinopie, 2019-08-18:
i see.. from what i found is different, the middle 3 numbers/letters is model code, i dont know maybe it depends on where market sales for that bike.. i know little about oem spec by market, i used to work as product development engineer in a car maker company..

edwinopie, 2019-08-18:
if i compare those file (same end 3 digit) using hex editor or winols, clearly it seen that they are identical.. not 100% of course,,

RyanHope, 2019-08-18:
yes the middle 3 roughly lines up with model code but the same ecus are used on CB and CBRs and those are clearly different models, what is the same is they run the same engine and thus use the same ecu hardware

edwinopie, 2019-08-18:
please take a look..
[image]

RyanHope, 2019-08-18:
What are you trying to show/prove here?

edwinopie, 2019-08-18:
a guide line for unknown ecu, in my opinion the last 3 digit is maybe the closest assumption to be a starting point. so i think it will be good to add a parameter to input last 3 digit (if the user has no clue about start read address) and connect to a known database to get a start reading address.. this is only an input for HondaECU to make it more adaptable to unknown ecu..

RyanHope, 2019-08-18:
None of what you just said address my question about what you were trying to imply with that screenshot?

edwinopie, 2019-08-18:
the screenshot shows it look identical, so i guess the start reading address could be the same because those files has the same last 3 digits.. i dont know, correct me if im wrong..

RyanHope, 2019-08-18:
The start of those files all look the same cause the hardware is the same. And, all the examples you posted were from a similar year/generation so they were running the same OS version (i.e. same version of PGM-FI). The files are still very different if you look beyond the first few bytes in your screenshots... but those differences are all in maps and tables mostly.

edwinopie, 2019-08-18:
noted..

closed 2019-08-23 by edwinopie