MCU-Innovations/HondaECU #60: 38770-K56-M01

briantrepo, 2019-08-19:
Hai , does anyone here have the original RS150 38770-K56-M01 bin file.  please give it to me to fix it again.  thanks

RyanHope, 2019-08-19:
The issue tracker is not a big request site...

closed 2019-08-19 by RyanHope

briantrepo, 2019-08-19:
I'm sorry, but I'm having a problem reading ecu but the results are bad,
and I want to get the best results.
 Once again, I'm sorry, thank you

RyanHope, 2019-08-19:
Thats probably because it is running a buggy version of PGM-FI and the read is corrupted. It has been mentioned many times in these threads that many/most of the new 38770-K**-*** ecus must be read from the backdoor to get a good read.

briantrepo, 2019-08-19:
I would love to see a video of how you read through the back, so I know
it.  You can share it via Facebook or anywhere.  please make a video for me
and for others.  thank you.

RyanHope, 2019-08-19:
There are some tips about it on my facebook page. But basically, depot the ecu, find all the test pads, figure out the Tx/Rx/Vcc/Gnd pins, connect them to a ftdi breakout board and then use Renesas FDT to do the read.

briantrepo, 2019-08-19:
Ok , i want to know when you start running the FDT application until it's
finished.
 Can you make it  I will wait for your video, that's all I want from you

RyanHope, 2019-08-19:
You are asking a lot of me man, this is a free project...

briantrepo, 2019-08-19:
I'm sorry if you ask too many questions, sorry my friend just looking for a
solution.

RyanHope, 2019-08-19:
See #61