MCU-Innovations/HondaECU #61: 38770-K**-*** ECUs

RyanHope, 2019-08-19:
The constant questions about these ECUs are getting time consuming and tiring. Since this project has been public I have barely received enough donations to purchase a single used ecu for R&D and have sacrificed 5 of my own 38770-K**-*** ECUs to get clean reads and posted the bins. Only 1 person has shared a bin from a backdoor read they have done on their own and I know a few people have done backdoor reads since I have explained the process and are not sharing. I know there is a competitive market tuning the bikes/scooters that these ECUs are found in, but I can't hand hold everyone though the backdoor process to not get anything in return. If there is an ECU you can't get a good read of you can send it to me to backdoor and I will make the bin public.

uszzz, 2019-08-19:
What is a preferred way of sharing bin files? Github? Email?

RyanHope, 2019-08-19:
The preferred way would be via a github pull request. Fork the project, add the bin maintaining the folder/file naming conventions, and then once that is committed in your in your fork do a pull request on my repo and I can merge the change. Then you would show up as a contributor to the project. If you dont want to do that email it to me. Please include the ecu part number, model name and the ecmid if possible.