MCU-Innovations/HondaECU #62: EEPROM Write/interrupted

prostreetprotege, 2019-08-20:
Hey Ryan
i'm test EEPROM Function on V3.3.0
Read is OK
Format is OK
Write is interrupted
Bike have a HISS Show and can't Start

Use V3.3.0 and D.I.Y Cable

prostreetprotege, 2019-08-20:
[image]

RyanHope, 2019-08-20:
Are you testing on the bench or on the bike? I have not been able to do
much testing with bikes that actually have HISS. Once of my regular testers
has noted some similar issues.

prostreetprotege, 2019-08-20:
I test on the bike.
I delete hiss location in bin and flash to bike.
My bike can Start again

uszzz, 2019-08-20:
Can you advice how to turn HISS off in bin? Thanks in advance.

prostreetprotege, 2019-08-21:
I can't tell the position clearly.
Because each ecu is different
In cbr500, it's around the ecmid

uszzz, 2019-08-21:
Then what byte(s) to look for?

prostreetprotege, 2019-08-22:
You must find ecmid location first.
And hiss is about 2 lines before ecmid

RyanHope, 2019-08-22:
not on all ecus

elmikel147, 2019-08-22:
good morning to all of you. what does ecmid mean? thanks to this wonderful forum.

closed 2019-08-27 by prostreetprotege